victronenergy/dynamic-ess #88: [Bug]: No colour one some hours in energy graph

KajLehtinen, 2023-10-01:
### Contact Details

kaj261@gmail.com

### VRM portal ID

d41243cd7f0b

### Country / region

Sweden - SE3 (se-3)

### B max

20

### TB max

5

### FB max

9

### TG max

2

### FG max

11

### Battery costs

0.0001

### Buy price

(p+0.027692+0.041281)*1.25

### Sell price

(p-0.02527+0.00)*1.25

### feed-in possible

yes

### feed-in control

yes

### Version

0.1.6

### What happened?

I got some hours that are white in the energy graph, see below

[image]

There is a white unlabeled section in the popup legend and only forecasts for PC yield and consumption is shown during some hours. The white section seems to equal the consumption forecast in the graph.

/Kaj  

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

KajLehtinen, 2023-10-01:
Looking at the Overviwe graph the system SoC is flat during most of the same time
[image]

And the Dynamic ESS schedule is locked to 63%

[image]

dirkjanfaber, 2023-10-03:
The white value is essentially the part below both of the lines (consumption and pv yield). The part between should be filled as well, but when price is below zero, things become special. We are looking into improving the graph in such cases.

closed 2024-02-26 by dirkjanfaber