victronenergy/dynamic-ess #89: [VRM - Bug]: day ahead prices are flat

Marous23, 2023-10-03:
### Contact Details

marek.vagner@gmail.com

### VRM portal ID

c0619ab1ea67

### Country / region

Czech Republic (cz)

### B max

12

### TB max

12

### FB max

12

### TG max

17.25

### FG max

17.25

### Battery costs

0.04

### Buy price

(p+0.012+0.032)*1.21

### Sell price

(p-0.02+0.032)*1.21

### feed-in possible

yes

### feed-in control

yes

### What happened?

I am back testing the Dynamic ESS. This time the VRM implementation. This morning I wanted to check the graphs and it is strange but the prices are not loaded correctly. The graph is flat, totally different than the real one. The flat price was visible yesterday at evening too.  I am attaching a screen took on my phone.
[image]

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

dirkjanfaber, 2023-10-03:
You have set the Dynamic ESS to be off right now. 

[image]

So set that to auto and it should become functional again.

Marous23, 2023-10-03:
Yeah, I switched it off because I saw the graph with flat price, obviously broken. So I tought the plan based on broken price wont be good too. And it was showing just „Loading“.
Now I switched back to „auto“. But the price is still flat.
[image]

Marous23, 2023-10-03:
I see the current plan as „send all to grid and keep the battery flat“. 
[image]
That does not make sense in context the prices are currently almost zero.
[image]
That confirms the loading of prices is broken and the derived plan too.

dirkjanfaber, 2023-10-03:
Indeed. Can now confirm that there is a problem with the day ahead prices being fetched by the API. We are working on it.

Marous23, 2023-10-03:
Perfect. Thanks. I believe it started already at yesterday evening.

LecramRed, 2023-10-03:
Same here. Good to read that it's a known issue. The Node Red version works well.

dirkjanfaber, 2023-10-03:
It should be back online now. We will be adding more monitoring to this part of the system. But as we are partly depending on third-party services (e.g. entsoe for the pricing), there is no guarantee that it won't happen ever again.

closed 2023-10-03 by dirkjanfaber

Marous23, 2023-10-03:
Perfect. Thank you. I will try later today to switch to auto again. I understand it is problematic to rely on entsoe API. Any small change at their side and you are without data. If they are doing so it's inconsiderate of them.
Anyway thanks for your work. I am looking forward to later versions. I already switched to spot prices recently and I consider this kind of system like a must have. My first idea was to implement some light logic in Loxone myself. But it would take ages to test the ideas in real life and why to invent another wheel. So I am very happy that this is going to be a native feature of Victron system.