victronenergy/dynamic-ess #91: [Bug]: Dead-Zone in planning

SimonYoungtree, 2023-10-04:
### Contact Details

siempoot@gmail.com

### VRM portal ID

c0619ab31cc4

### Country / region

Netherlands (nl)

### B max

40

### TB max

6

### FB max

6

### TG max

16

### FG max

16

### Battery costs

0.027

### Buy price

(p+0.13)*1.21+0.02

### Sell price

(p+0.13)*1.21-0.02

### feed-in possible

yes

### feed-in control

None

### Version

0.1.6

### What happened?

The planning shows a dead-zone. I do not think this is correct.

[image]

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

SimonYoungtree, 2023-10-05:
After a number of hours (I do not know exactly when), some consumption was planned again. Perhaps history on VRM could not be compiled for some reason?

andreashackel, 2023-10-15:
I could also observe this problem in my system environment (Node Red V.0.1.7). In the last 8 days there was only one day without zero data in the consumption forecast. Otherwise the problem occurs after 13:00 and is solved by a recalculation after 4 to 8 hours. Today (20231015) I was able to take a closer look.
As usual, after 13:00 the consumption forecast changed a lot, but today it did not go to zero, but to a very small value (< 10Wh). Since I have a base load of about 250 Wh in my house, this forecast is not plausible. (Is it possible that only the grid consumption is taken into account and not the battery consumption?) A recalculation around 14:30 corrected the planned consumption to plausible values around 800 Wh.

20231015 12:52 (On this day in the period from 18:00 to 24:00.)
[image]
20231015 13:12
[image]
20231015 14:25
[image]

SamuelBrucksch, 2023-11-07:
I have the same issue on VRM beta:
[image]

dirkjanfaber, 2024-02-26:
Closing this issue as it has been open for too long; new releases have been out. If it still occurs with the latest release, please open a new issue.

closed 2024-02-26 by dirkjanfaber