victronenergy/dynamic-ess #92: [Bug]: Target SOC 20% -> 10% is not being executed. No discharging

SimonYoungtree, 2023-10-06:
### Contact Details

siempoot@gmail.com

### VRM portal ID

c0619ab31cc4

### Country / region

Netherlands (nl)

### B max

40

### TB max

6

### FB max

6

### TG max

16

### FG max

16

### Battery costs

0.027

### Buy price

(p+0.13)*1.21+0.02

### Sell price

(p+0.13)*1.21-0.02

### feed-in possible

yes

### feed-in control

None

### Version

0.1.6 - OS version 3.10-(with bugfix)

### What happened?

According schedule the Target SOC should go to 10% (now 20%), however we are now 45 minutes into that hour 8:45 and no discharging is taking place. Therefore DESS will not meet its plan

[image]
[image]

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

dirkjanfaber, 2023-10-06:
Your minimum SOC right now is 20%. So it won't go below that, even if it would like to.

[image]

closed 2023-10-06 by dirkjanfaber

SimonYoungtree, 2023-10-06:
Where does this 20% come from? My min SOC is 10%:

[image]

dirkjanfaber, 2023-10-06:
If you check under the _Dynamic ESS_ option (last item under _ESS_). What does it say there as minimum SOC?

SimonYoungtree, 2023-10-06:
There is nothing shown about min. SOC. See new picture on GitHub.

dirkjanfaber, 2023-10-06:
Oh yeah, the console is not showing that min soc when running the Node-RED mode. 

You can see the value with this flow:
```
[{"id":"d8409391cd8338bf","type":"victron-input-custom","z":"15669bac1d63a856","service":"com.victronenergy.settings","path":"/Settings/DynamicEss/MinSoc","serviceObj":{"service":"com.victronenergy.settings","name":"com.victronenergy.settings"},"pathObj":{"path":"/Settings/DynamicEss/MinSoc","name":"/Settings/DynamicEss/MinSoc","type":"number"},"name":"","onlyChanges":false,"x":490,"y":1300,"wires":[["72b29acded57de34"]]},{"id":"72b29acded57de34","type":"debug","z":"15669bac1d63a856","name":"MinSoc","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"payload","targetType":"msg","statusVal":"","statusType":"auto","x":840,"y":1300,"wires":[]}]
```

SimonYoungtree, 2023-10-06:
OK, used custom control to set it to 10%. Thanks