victronenergy/dynamic-ess #93: [REOPEN] [Bug]: Target SOC 20% -> 10% is not being executed. No discharging #92

SimonYoungtree, 2023-10-06:
### Country / region

None

### feed-in possible

None

### feed-in control

None

### What happened?

See #92

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

dirkjanfaber, 2023-10-06:
I don't see a reason to re-open the other issue. As far as I can see it has been rightfully closed. And I will still respond to comments posted in that issue (even while it is closed). I'll re-open it when I see a good reason to do so. Really no need to open an extra issue for this.

Btw it is not my intention for being blunt or short in my answers. I get a lot of questions and distractions while trying to go through all of them. All time I spend on answering is time that I can't spend on actually programming and fixing issues.

SimonYoungtree, 2023-10-06:
[image]

SimonYoungtree, 2023-10-06:
To avoid waisting your time: I do not file things if there is no legitimate
reason. (Being a software professional myself, being able to distinguish
right from possibly wrong). I am just trying to help.