victronenergy/dynamic-ess #94: PV-Ladegerät speist über Multiplus 2 nicht mehr ins Netz[VRM - Bug]:

PV-Stefan, 2023-10-06:
### Contact Details

S.essingholt@gmail.com

### VRM portal ID

c0619ab1d3c9

### Country / region

Germany (de)

### B max

26

### TB max

3.7

### FB max

3.7

### TG max

21

### FG max

21

### Battery costs

0,03

### Buy price

P+0,02+0,12)×1,19

### feed-in possible

None

### feed-in control

None

### What happened?

Batterie ist voll, es wird aber nichts mehr über das dc pv Ladegerät mit mutiplus 2 eingespeist 

Ohne dess wird eingespeist 

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

dirkjanfaber, 2023-11-03:
Sorry that it took so long to respond to this issue, which I had overlooked. If the issue is still valid, can you please rephrase the question in English as I fail to understand what the exact issue is that you report? Mein Deutsch ist leider nicht so gut. 

Also note that we did release version 3.20~17 (candidate) today, which does fix some issues regarding Dynamic ESS. And it also offers some [restrictions](https://github.com/victronenergy/dynamic-ess#restrictions), which are interesting for the German market.

PV-Stefan, 2023-11-18:
Thank you for your answer.
Realy great Job.
I wish tre dess will work soon.
I try it sometimes,  but i prefare the normal ess with manual setup,  it
works bester at the Moment.
Is this case in the picture the grid price is realy low, baut the System
will Not load the battery from grid. Even there is planed solar to grid.
I prefare all solar to battery and get a higher state. Eventuell the
battery should load the cheap grid to reach a maximum of the batterie.
The planed baterie maximum is 58% it should be 100. So the System should
load the cheap grid up to 40%.