victronenergy/dynamic-ess #99: [VRM - Bug]: still ilogical planning for charging/feeding/consumption

khostri, 2023-10-11:
### Contact Details

jan.slaby@outlook.com

### VRM portal ID

210204

### Country / region

Czech Republic (cz)

### B max

21

### TB max

10

### FB max

10

### TG max

15

### FG max

9.9

### Battery costs

0.04

### Buy price

p+0.186164629

### Sell price

p-0.02214759

### feed-in possible

yes

### feed-in control

yes

### What happened?

As reported several times, in some algo planning is totaly wrong in my oppinion for some days.
I have DESS turned on for some time, switched from NodeRED to VRM version. I am usually ok with planning (ok I mean I can live with it but not totaly exited, as DESS is charging discharging in one hour many times, compensate for target SOC doing full power discharge of energy it was charged one minute ago etc).
But today planning seem totaly off today that I even switched DESS off.
There is solar yield expected, battery is low on 28 percent, sell prices are low nearly on zewro and system is not planning charging battery but feeding to grid for nearly zero. Even from batery as soc is 28 and planned was 25...
It does not make sense at all. I dont expect to plan for 100 soc (although wuth standard operation mode without DESS I would go for that), but not plan charging at all?
Tried to leave DESS turned on and set Active soc to 85%, but in that case DESS started to feed from grid as crazy.

[image]
[image]

### When did it happen?

11/10/2023

dirkjanfaber, 2023-10-11:
I see that your system is still running 3.10. I do advise you to go to 3.11, as that does fix some things regarding the Dynamic ESS handling of the system. Mainly the quickly charging and discharging should be fixed by that. 

I definitely agree that sometimes the system does things that seem illogical at first sight. If you look deeper into it, it turns out that it does make sense on how the system plans. For example, you have set your battery cost to € 0.05 / kWh (not in this report, but according to the logging of the API you do have it set to that).

So when the sun shines there are 2 possible options for the system to choose from: 
- feed into the grid directly (for sell price/kWh). For example for € 0.03 / kWh.
- charge the battery and sell later (for the sell price at that later moment/kWh). For example for € 0.11 / kWh.

Keep in mind that the second option adds € 0.10 / kWh of cost to the equation. First it costs € 0.05 / kWh for charging your battery from the solar and later it adds another € 0.05 / kWh for discharging it to the grid. 

So in the above example it yields more money to discharge now (for € 0.03/kWh)  instead of later today (for € 0.11 - € 0.10 = € 0.01/kWh).
 
There are places where the system can be improved and we are working on that. E.g. if you use more consumption then forecasted, it would be cheaper to use the €0.10 / kWh solar (from the battery) then using it from the grid.

I will close this issue as the taken choices are not considered a bug. If you want the system to take other decisions, you might want to re-evaluate the set prices in the configuration. Lowering the battery costs will increase the usage of the battery.

closed 2023-10-11 by dirkjanfaber

khostri, 2023-10-11:
Hi @dirkjanfaber I understand that. My point is, that for second option you mentioned, there is not only one stream, but 2l suboptions to be evaluated - your option that charging battery cost is there + discharge cost into grid.
But there is also evaluation, that there is only cost for battery charge. For dischargin, you are not comparing cost of discharge to grid vs sell to grid, but cost of discharging for consumption vs buying from grid to cover consumption I think. And as you see the difference of bying/selling price is high I thin that evaluation woudl lead to different planning :)