victronenergy/dynamic-ess #101: [Feature]: Show the savings by dynamic ESS

pokeplayer2, 2023-10-13:
### Description

You show today only costs:

[image]

### Suggested Solution

Show also the savings into the empty area ... my old costs for power from grid would be €0.39, but I save now money with the dynamic ESS solution → let's make it visual for the people ;-)

tuxedo0801, 2023-10-27:
I'm also confused by the negative values earnigs with and without D-ESS) ... Shouldn't earnings be positive and not negative? 

A negative earning is a kind of loss, or?

pokeplayer2, 2023-10-31:
Yes and the mouse over help message is not helpful.

jwillmer, 2024-10-03:
Where can I find this feature? I don't use NodeRed so I guess it is not available in the portal yet? I use HomeAssistant and would like to track how much cost savings I get from this system. Meaning utilizing my battery like this.