victronenergy/dynamic-ess #102: [Feature]: add a Shelly interface for a heating system / switches

pokeplayer2, 2023-10-13:
### Description

I like to switch on my water heating system when the price reached the daily lowest price, between 1 and 5 pm. When the consumption goes down to under x Watt, I switch it again off. 

[image]

### Suggested Solution

It's a start point to support a very simple heating system and  during the winter time you have two time periods which are from interest: 
[image]

In the night and during the day, and that is more as enough for a water heating system.

dirkjanfaber, 2023-10-17:
Adding such functionality is not yet on the shortlist, but meanwhile there is a trick with using the Node-RED flow to achieve this functionality. If you import [this](https://github.com/victronenergy/dynamic-ess/blob/faberd/extra-example/examples/nodered-ui-with-vrm-dess-with-average-price-switching.json) example and adjust for Shelly relays. 

Not too sure if a heater really likes to be switched by a relay though.

mman, 2023-10-17:
I am using Node RED and Shelly Pro 2 to turn on/off two phases of my old BMW EV charger, and it works great, pushing through not more than 12.5A per phase as indicated in the Shelly specs, and it gets to around 40 degrees celsius and did not switch itself off, so depending on power requirements of the heater, I have very good experience with the Shelly Pro for now!