victronenergy/dynamic-ess #103: [VRM - Bug]:

pokeplayer2, 2023-10-14:
### VRM portal ID

b827eb6ff17f

### Country / region

Germany (de)

### B max

26

### TB max

6.5

### FB max

7

### TG max

30

### FG max

30

### Battery costs

0,02

### Buy price

p+0.217+0.0161

### Sell price

0,115

### feed-in possible

yes

### feed-in control

yes

### What happened?

Battery discharge only with 2,5 kW if discharge is allowed, why we do not discharge with the need power? At this moment was 5,5KW needed and the rest comes from grid. :-( Grid is mostly expensive here and when the battery has enough power, why should we limit the discharge power?

### When did it happen?

14/10/2023 09:50

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

pokeplayer2, 2023-10-14:
[image]

KajLehtinen, 2023-10-14:
The way Dynamic ESS works currently is that it sets a target SoC and aims for that during the current hour, if the consumption is more than estimated it will consume from grid, and if its lower it will sell back to grid if allowed.

/Kaj

pokeplayer2, 2023-10-15:
I don't like it as it end in a situation where you buy power at totally wrong time. At that moment was the battery ~ 83 %, no need to buy power from grid.

pokeplayer2, 2023-10-15:
and again power from grid and battery over 80% 
[image]

pokeplayer2, 2023-10-15:
[image]
23:00 - 00:00 was grid used ... 15/10/2023

dirkjanfaber, 2023-10-16:
There was an inaccuracy in forecasts, so in hindsight the battery could have discharged more than it did. Note that hindsight clarity is always easy. Determining what to do based on the knowledge of the present is less simple.
At the moment we don't offer ‘course-correction’ logic somewhere within the hour. We are researching if there is a good way to do this, but we decided to put it to lower priority to focus on more pressing matters first. 
I don't minde leaving this issue open, but it won't get much further attention for a while.

pokeplayer2, 2023-10-16:
I have no problem with the long runner, ;-) important is that the solution is in future more agile and the limits have to be removed.

dirkjanfaber, 2023-11-28:
With the 0.1.10 version we've introduced restrictions that are followed during the scheduled hour. See the [restrictions](https://github.com/victronenergy/dynamic-ess#restrictions) section in the README.

At the moment the logic only sets the `2` or `0` value, If that is not enough, you can add some extra logic to the flow by injecting the needed restriction, based on the information available.