victronenergy/dynamic-ess #105: [VRM - Bug]: battery load  starts to early

pokeplayer2, 2023-10-16:
### VRM portal ID

b827eb6ff17f

### Country / region

Germany (de)

### B max

26

### TB max

6,5

### FB max

7

### TG max

30

### FG max

30

### Battery costs

0,02

### Buy price

p+0.217+0.0161

### Sell price

0,115

### feed-in possible

None

### feed-in control

None

### What happened?

Load battery at too high price 

### When did it happen?

16/10/2023 09:00

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

dirkjanfaber, 2023-10-16:
At the moment this is considered expected behavior (See the [manual](https://www.victronenergy.com/live/drafts:dynamic_ess#determining_the_target_soc)):
> Also keep in mind that, when doing a consumption forecast, the system does not known when during the hour that consumption will be done. If the forecast does not match the reality, Dynamic ESS does keep the planned battery usage stable while punishing the forecast inaccuracies on the grid (selling the excess to grid, buy the additional need from the grid). This can be less ideal, but it is an issue of the forecasts being inaccurate, nothing that the Dynamic ESS decided.

We are working on adding a way to disallow feeding from grid into the battery. That will likely solve this.

pokeplayer2, 2023-10-16:
I allow feeding from grid when the price is low and the battery is empty ;-)  in the morning hours it was cheap and during the day ... no problem to feed from grid. I look only at the money and at this moment it was terrible ... high price and battery had a good amount of capacity. Other solutions working with minimum hourly load which can be manipulated by the user. I start this week my heating system, so I know my daily needs of power is much higher as 28 days before ... the forecast will be wrong by default.

dirkjanfaber, 2023-11-03:
From the latest candidate release (3.20~17 and higher) it became possible to add restrictions (see https://github.com/victronenergy/dynamic-ess#restrictions). 
Can you switch to that and test? I'll close this bug for now. Please create a new one if the problem persists with that new version.

closed 2023-11-03 by dirkjanfaber