victronenergy/dynamic-ess #106: [VRM - Bug]: Status about dynamic ESS and before

pokeplayer2, 2023-10-17:
### VRM portal ID

b827eb6ff17f

### Country / region

Germany (de)

### B max

26

### TB max

6,5

### FB max

7

### TG max

30

### FG max

30

### Battery costs

0,02

### Buy price

p+0.217+0.0161

### Sell price

0,115

### feed-in possible

yes

### feed-in control

yes

### What happened?

Here is the status of the dynamic ESS solution before I switched on and when it is on.

### When did it happen?

13/10/2023 07:00

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

pokeplayer2, 2023-10-17:
09/1010/2023
[image]

10/10/2023
[image]

11/102023
[image]

12/10/2023
[image]

13/10/2023
[image]

14/10/2023
[image]

15/10/2023
[image]

16/10/2023
[image]

Overall, we buy too much from grid during the expensive time with dynamic ESS.  -> I will check another solution or go back to the old script.

dirkjanfaber, 2023-10-17:
This is confusing. Your overview has a lot more costs than we can deduce. Your grid readings wildly differ from what is shown and recorded in VRM. 
For example your grid numbers (as recorded for the 16th of October):
[image]

Which does not seem to match you graphs. From what measurements are your graphs coming from? If we check the VRM recorded measurements of L1, L2 and L3 from between 4:00 and 5:00, we are unable to match those to the 0.66 kWh grid usage your graph is showing:
[image]

You are of course free to use whatever system you want and works for you. If the calculations happen with wrong data and assumptions, it is not surprising that the wrong actions are taken. At the moment that looks specific for your site, so it is interesting to check where the cause of getting the wrong values lies. And you are sure that this started when you enabled DESS? Or did more things change on your setup?

Also, which "old script" are you referring to?

pokeplayer2, 2023-10-17:
Thx for your feedback, since Friday is the API to get the grid data down for my power provider :-(  
[image]
so I can't trust them and VRM is correct. That is terrible, just when I start to test dynamic ESS was the API "down". All local screenshots are correct, but the screenshots from grid power company a wrong from 13/10/2023.

dirkjanfaber, 2023-10-17:
Admittedly I had not expected that, but at least it does explain the differences. Hope that your provider fixes their API issues soon, so we can really compare the numbers. 
If you don't mind I'll close this issue now. Please do create a new one when their API works again and the numbers don't match.

closed 2023-10-17 by dirkjanfaber