victronenergy/dynamic-ess #108: [Bug]: sudden error: Dynamic ESS: Battery level unknown

SimonYoungtree, 2023-10-19:
### Contact Details

siempoot@gmail.com

### VRM portal ID

c0619ab31cc4

### Country / region

Netherlands (nl)

### B max

42

### TB max

6

### FB max

6

### TG max

16

### FG max

16

### Battery costs

0.025

### Buy price

(p+0.13)*1.21+0.02

### Sell price

(p+0.13)*1.21-0.02

### feed-in possible

yes

### feed-in control

None

### Version

0.1.7 and Venus-OD 3.11

### What happened?

I get the error code on the VRM site : Dynamic ESS error code: Dynamic ESS: Battery level unknown. VRM shows the correct percentage and Watts and current are moving. Same on the Cerbo. Also on NODE-Red everything show connected. How can this happen? Things seem to be working normal.

[image]

### When did it happen?

19/10/2023 12:27  (Visible in Alarm LOGS)

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

h2009, 2023-10-19:
I’m getting the same issue. Just started. 
[image]

dirkjanfaber, 2023-10-19:
This should not be an alarm. We are fixing that. You can ignore it.

AlexImb, 2023-10-19:
It should be now fixed.

SimonYoungtree, 2023-10-19:
Yes, it is gone.

dirkjanfaber, 2023-10-19:
Closing this issue.

closed 2023-10-19 by dirkjanfaber