victronenergy/dynamic-ess #109: [VRM - Bug]:

pokeplayer2, 2023-10-19:
### VRM portal ID

b827eb6ff17f

### Country / region

Germany (de)

### B max

26

### TB max

6,5

### FB max

7

### TG max

30

### FG max

30

### Battery costs

0,02

### Buy price

p+0.21*1.19

### Sell price

0,115

### feed-in possible

yes

### feed-in control

yes

### What happened?

Buy price is wrong

### When did it happen?

19/10/203 20:25

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

closed 2023-10-20 by dirkjanfaber

pokeplayer2, 2023-10-20:
Argh, sorry and thx!

pokeplayer2, 2023-10-20:
How fast is the chart re-created after config changes? It's very hard to check the configuration when the changes are not immediately visible ...

dirkjanfaber, 2023-10-20:
Afaik historic graphs are not altered, so left of the vertical line keeps the calculations from that moment. The rest of the graph (right of the vertical "now" line) should be recalculated direct after changing the configuration. But I might be wrong on that, in which case it is max 1 hour.

vbdaniel4, 2024-02-13:
Formula ist incorrect. Should be (in your case):
(p*1.19)+0.21