victronenergy/dynamic-ess #110: [VRM - Bug]: no efficient battery usage

pokeplayer2, 2023-10-23:
### VRM portal ID

b827eb6ff17f

### Country / region

Germany (de)

### B max

26

### TB max

6,5

### FB max

7

### TG max

30

### FG max

30

### Battery costs

0,02

### Buy price

(p+0.21)*1.19

### Sell price

0,115

### feed-in possible

None

### feed-in control

None

### What happened?

A bug happened!

### When did it happen?

DD/MM/YYYY HH:MM

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

tuxedo0801, 2023-10-27:
I can confirm this from another point of view:

Weather is bad, battery is low, but there is some PV power available. Instead of charging all the power into the battery, Dynamic ESS is feeding some of it into the grid. 
With a fixed sell-rate of 7.5cent per kWh this makes absolutely no sense.

[image]
[image]

VRM portal id: 48e7da859663
feed-in possible: Yes
feed-in control: Yes
Screenshots taken at 2023-10-27 09:30am GMT+2

pokeplayer2, 2023-10-29:
[image]
Grid is very cheap and dynamic ESS like to use batteries ... WHY ?

tuxedo0801, 2023-10-30:
I'm curios how victron further develops this function. 
I think it's not everytime easy to understand DESS's decisions, as you also need to take into account how the PV and consumption forecast looks like. 

I'm now trying to go for a different approach: In summer-time, I have enough PV to survive without any grid. In winter time, I know more or less precisely how much energy I need to drive our house and heating. So I do not need a forecast that get's confused when things like heatpump or EV starts consuming. I know, without EV, I need between 10 and 20kWh per day. 
My battery has 30kWh. So its enough for at least 24hrs. 

With that I can calculate, depending on the PV forecast, how much enery I need from the grid. And with an hourly grid price, I can calculkate when it makes sense to charge my battery to which SOC. 

As you EVs are parking the whole night in the garage, they can wait for a low price and start charging at this time to a minimum that is required to survive the next 24hrs. 

I think this will do it, and will be very transparent and understandable compared to DESS decisions :-) Will see how this works out for me. Just started to develop this solution.

Je68le, 2023-10-30:
> I think this will do it, and will be very transparent and understandable compared to DESS decisions :-) Will see how this works out for me. Just started to develop this solution.

Hello Alex,
Good to know that you are looking for a transparent solution, because I have the same problem as you, finding an easy way to combine the solar forecast with knowing the amount of electricity required per day. Please let me know if I can help you provide an additional testing environment ;-)
Jens

pokeplayer2, 2023-10-31:
No battery load today: 
[image]
SOC is too low: 
[image]

before I go to bed it was planned to load the battery between 2 - 4 o'clock and now noting :-/

321kontor, 2023-11-09:
Same here. I noticed it several times now, that charging was scheduled in the most expensive slot. And the battery was at 75% already. My second point is charging when sun is shining. In my opinion it should always be better (at least in germany) to charge with maximum kw. Energy out of battery is always cheaper than from grid. I had it several times, that battery was charged with 300w-500w, when 1700w was easy possible.

[image]
[image]

321kontor, 2023-11-09:
btw charging now!?
[image]

tuxedo0801, 2023-11-09:
same here: 
[image]

Why would I feed to grid if battery has a SOC of ~41% (30kWh total capacity)... Turned off DESS once again... 

I doubt that people will use DESS if it's not explainable why the system behaves like this.

tuxedo0801, 2023-11-09:
@Je68le 
working on a simple solution. Current status:
If current price drops below daily average, I signal via "SnmartGrid" my heating system that price is low and that it's a good idea to heat or prepare warmwater NOW. Next step could be: if battery SOC is that low that it will not survive the next hours (or night or day ... not that clear for now): let the battery charge from grid if price is also below daily average. The tricky part is: how to calculate if it will not survive. I think I will be based on a specified SOC level as it's not really reliable estimateable when heating with heatpump or EV charger will turn on. 
I also doubt that my solution will be 1:1 applicable for others. But maybe it's kind of inspiring.

pokeplayer2, 2023-11-16:
[image]
ESS stopps from my point of view, one hour to early to use the battery:
- Grid is at 9 pm at 35 cent which would be ok for today, but ESS stops at 8 pm 36 cent
- we have still over 60 % into the battery

new VRM ID c0619ab4ae0e

dirkjanfaber, 2024-02-26:
Can you try again with running version 0.0.14 with firmware 3.30~7 (or above when released)? This adds extra strategies for not blaming the wrong forecasts on the grid all of the time. 

Closing this issue. Please re-open a new issue if the problem persists.

closed 2024-02-26 by dirkjanfaber