victronenergy/dynamic-ess #111: [Bug]: Small peaks at start of an hour

MichaKersloot, 2023-10-23:
### VRM portal ID

c0619ab410ae

### Country / region

Netherlands (nl)

### B max

9.8

### TB max

3

### FB max

3

### TG max

4.1

### FG max

4.1

### feed-in possible

yes

### feed-in control

yes

### Version

0.1.7

### What happened?

At some hours the SOC setting changes after the first minute of the hour. Not much for me, but maybe this points to a bug somewhere.

[image]

Furthermore a strange glitch at 12:49 where the SOC goes down from 28% to 22%  in the last 10 minutes while the next hour is already calculated to be 34%. This results in selling at almost the lowest price for the day.

### When did it happen?

23/10/2023 11:59

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

dirkjanfaber, 2023-10-25:
The small peaks at the start of the hour are normal behaviour. I believe it is the offset between the forecast of the previous hour and the actual data. So not much we can do about that (so that is no bug).

The one at 12.49 might be that the new day ahead pricing came in and the calculation was redone, based on the extra knowledge. But I need to have that confirmed by a colleague.

MichaKersloot, 2023-10-25:
This could be, and as long as we have 'saldering' in the Netherlands it's no problem. But when that stops, it could be we end up paying for taxes only on the imported electricity and don't get it back for exported electricity. We then need to pay extra taxes for the electricity that is used to re-import the electricity that has been exported unneeded just before.

dirkjanfaber, 2023-10-25:
Got confirmed that my hunch about the new price info coming in at that 12.49 point.

For the German market we are already adding some extra features, on which the Dutch can ride along once needed. By the time the saldering stops, it should not be a problem.

Closing this issue as it is all expected behaviour.