victronenergy/dynamic-ess #113: [Feature]: When target is 98 or 99 then make target 100% and keep charging at CVL (and respect CCL) to allow for balancing

SimonYoungtree, 2023-10-30:
### Contact Details

siempoot@gmail.com

### Description

When target is 98 or 99 then make target 100% and keep charging at CVL (and respect CCL) to allow for balancing.

I could inject the 100% myself (via Node Red) when the target is 98% or 99% to override things. However the charging will not keep pushing to the CVL the way things have been implemented at the moment. So  currently this workaround does unfortunately not work.

### Suggested Solution

When target is 98 or 99 then make target 100% and keep charging at CVL (and respect CCL) to allow for balancing.

### Alternatives

Until solved: manual intervention (disable DESS and set schedule to 100% for a number of hours) once per month to do some balancing.

MichaKersloot, 2023-11-01:
Just a question, is your system only balancing on 100% SOC? My pylontechs seem to balance when one of the cells is over 3,45V independent of SOC.  Furthermore there are settings about absorption, float and sustain in veconfig. Maybe that is the place you can finetune the settings to your need.

SimonYoungtree, 2023-11-01:
About the settings in veconfig: they are ignored when using DVCC. Also PV
charger uses DVCC settings.

MichaKersloot, 2023-11-01:
If you want to control this yourself instead of the BMS (which sets the CVL, CCL and DCL, I presume). You are maybe better off not connecting the BMS to the victron and use a shunt for SOC monitoring?

SimonYoungtree, 2023-11-01:
I just want the charger to do its work: charge up to CVL

SimonYoungtree, 2023-11-02:
I will build a workaround with node red: every hour I will check: when DESS has a target of 98% or 99%, I will turn of DESS and switch back to ESS for 55 minutes (or so) while injecting a charge schedule for one hour to 100%.
After 55 minutes, I will turn DESS on again to allow it to refresh its calculation. And then it repeats itself.

dirkjanfaber, 2024-03-18:
You could also add a function in just before the 'split schedule 0' function that contains:
```
if (msg.soc >= 98) {
    msg.soc = 100
}
return msg;
```

closed 2024-03-18 by dirkjanfaber

software-testing-professional, 2024-03-25:
Hi guys, 

I'm a bit late and just found this issue closed. 

@dirkjanfaber Can you please reconsider this issue, and maybe reopen it?

I have a ESS with 3x MultiPlus2 and an AutarcTech 11.2kWh battery. 
This battery has an active balancer (AutarcTech Joulie-16) - and it's always balancing, and not only at 100%. 

BUT: 
My BMS calibrates the 100% SoC only when: 
* at 54.8V  and
* at no charge for at least 5min

So if my battery never reaches 100% SoC for quite a while, it's not good for calibration. 

Could you please add some logic to have the battery fully charged at least once a week or so?

Thank you, 
Michael