victronenergy/dynamic-ess #114: [VRM - Bug]: SOC too high, against forecast usage

h2009, 2023-10-30:
### Contact Details

layth@layth.co.uk

### VRM portal ID

48e7da8580e7

### Country / region

None

### B max

76.8

### TB max

24

### FB max

18

### FG max

unlimited

### Battery costs

0

### Buy price

.075 fixed

### feed-in possible

no

### feed-in control

None

### What happened?

DESS, is charging the battery too much each day leaving excessive SOC in the battery. 
Forecast consumption seems too high as well but cannot see why the forecast is higher than expected. 

### When did it happen?

All the time

pokeplayer2, 2023-10-30:
For me is always too low ;-) ... I like to have energy into the battery for the case that the grid is damaged.

MichaKersloot, 2023-11-01:
Look at the settings -> ESS -> mode. If this is set to Optimized (with batterylife) that could be the cause. The results are better when set to Optimized (without batterylife)

h2009, 2023-11-01:
> Look at the settings -> ESS -> mode. If this is set to Optimized (with batterylife) that could be the cause. The results are better when set to Optimized (without batterylife)

Afraid this isn't the case. 
Most days I end up with SOC of 50% left.

h2009, 2023-11-01:
Hi
No the SOC is too high at the end of each day. So the DESS is charging the batteries overnight by too much. 
Battery is not enabled. 
This issue started when the mini and max SOC for DESS was removed.

pokeplayer2, 2023-11-02:
I update yesterday to the latest Venus version and today is SOC also to high on my site:

[image]
Min SOC is 10% and during the night is power cheap ... VRM portal ID  b827eb6ff17f:
[image]
it is normally not needed to use power from grid in this case.

closed 2024-02-26 by dirkjanfaber