victronenergy/dynamic-ess #115: [VRM - Bug]: sell to grid has a to high priority

pokeplayer2, 2023-11-04:
### Country / region

None

### feed-in possible

None

### feed-in control

None

### What happened?

A bug happened!

### When did it happen?

DD/MM/YYYY HH:MM

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

schwede70, 2023-11-10:
Same behavior from my system. Priority for an ESS system should still remain in buffering solar energy in battery instead of selling to grid. Sell price was defined with 8ct/kWh (true selling value). I now reduced to 1ct/kWh to enforce the charging of the battery instead of selling to the grid - Just to try it out. Now the system is charging with 70% of solar power and selling 30% to grid. Behavior is not understandable.

BR Dirk

dirkjanfaber, 2023-11-10:
We are aware of this and working on it. Priority will likely become based on the cheapest price between storing it into the battery vs buying it from the grid.

dirkjanfaber, 2023-11-28:
With version 0.1.10 we've largely solved this by adding restrictions per scheduled timeslot. These are automatically set to the most useful restriction, mainly depending on buy / sell price.  Currently there are the following options:

- `0` = No restrictions
- `1` = No exporting from battery to grid
- `2` = No importing from grid to battery

The global restrictions still have persistence above these settings. Also see [here](https://github.com/victronenergy/dynamic-ess#restrictions).

With the Node-RED flow, you can always add extra logic to the flow, if you need more or less strict restrictions during specific hours.

schwede70, 2023-12-12:
Hi,
Is there any further update on this topic? Last days my system was still selling solar power to the grid instead of charging the battery. Do I have an influence on that behavoiur while using the "betavrm" with V3.20~18 on my Cerbo GX device?
BR

schwede70, 2024-01-09:
Hi, I'm still stuggeling with the behavior of charging the battery from solar power. 
In the example below (screenshot from today), I have a battery SOC of around 88% (discharge is only allowed until 75%), sun is shining and the solar power is higher than my consumption, but the system is not charging the battery, but selling for 8cent/kWh to the grid. Is this a feature (intended behaviour)? In my opinion, the system should charge the battery first before selling to the grid (basic strategy of an ESS).

[image]

dirkjanfaber, 2024-01-09:
At the moment that is still not covered by the system. Priorities have been slightly different last weeks, but we are currently working on solving this issue. I expect it to be added to a candidate release before the end of the month.

Edrettikketet, 2024-02-16:
I noticed the same problem.  
My system is always pumping the solar energy straight into the grid, while I would rather prefer to store it in the batteries (until their full) and prefer usage of it myself.  It seems to me that pumping it to the grid is only profitable when the grid price is high enough, but preferring the current from the battery seems to me preferable in the peak hours as well.