victronenergy/dynamic-ess #116: [VRM - Bug]: change and discharge battery over longer time

pokeplayer2, 2023-11-04:
### VRM portal ID

b827eb6ff17f

### Country / region

Germany (de)

### B max

26

### TB max

6,5

### FB max

7

### TG max

30

### FG max

30

### Battery costs

0,02

### Buy price

(p+0.21)*1.19

### Sell price

0,115

### feed-in possible

yes

### feed-in control

yes

### What happened?

The battery is for over an hour in a pulse mode and change and discharge ... that will kill the battery over time.

### When did it happen?

04/11/2023 21:58

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

pokeplayer2, 2023-11-05:
Same issue also today: 
[image]

dirkjanfaber, 2023-11-06:
We've been looking into this. 

By the looks of it, you are using an unsupported battery (this is an observation, not an explanation). The battery SOC seems to jump between 43.8% and 44.4%, while the target is 44%.
That is why it jumps like that.
There is no gradual changing from 43.8 to 44.4, it jumps there directly. Probably a mathematical translation done in the 3rd party serial-bms driver.

At the moment we are using a (hardcoded) hysteresis of 1%, which we can probably make configurable for the situations where it is not sufficient. But that will wait until the next release. I advise you to switch the firmware update feed to be using the latest release candidate instead of the latest release. We will first roll out changes on candidate before the end up in the release firmware.

Out of interest, what type of battery are you using?

pokeplayer2, 2023-11-06:
I got it, thx  -> release candidate is activated. I use two batteries DIY 13 kWh with JK-BMS  + Aggregate BMS.

[image]
A new parameter with configurable hysteresis would be good and I would allow 5 - 10 %  ;-) and plan is that the battery will grow to 39 or 52 kWh.

pokeplayer2, 2023-11-06:
Hm ... is a update from 3.12 is at the moment not possible?

[image]

dirkjanfaber, 2023-11-06:
Thanks for all the info.

AFAIK, we are not experiencing troubles right now. 
[image]

Btw, the cause turns out not be the hysteresis, but is slightly different. The code checks for the exact number 44.0% to be hit before going into idle mode. Yours jumps over that value. We need to adjust that piece of the code.

pokeplayer2, 2023-11-06:
Hm, thx I will check later again ... currently is an update not possible for me. Same error as before.

mpvader, 2023-11-06:
You are using a RaspberryPi? There are issues with that, see modifications page.

pokeplayer2, 2023-11-06:
Ok, then I have to wait ... my wife will kill me if I invest this mouth again into solar system .. ;-)

pokeplayer2, 2023-11-06:
Ok ok, Cerbo GX is ordered ... I do not like to lose time here.

closed 2023-11-10 by jhofstee