victronenergy/dynamic-ess #117: [VRM - Bug]: Grid feed from battery despite deactivation in setup

schwede70, 2023-11-10:
### VRM portal ID

48e7dae08f17

### Country / region

Germany (de)

### B max

7

### TB max

3

### FB max

3

### TG max

30

### FG max

30

### Battery costs

0.05

### Buy price

(p+0.1178+0.0192)*1.19

### Sell price

fixed 0.082

### feed-in possible

yes

### feed-in control

yes

### What happened?

Monitored several phases of feed the grid from battery despite deactivation in setup of the dynamic ESS. 

[image]
[image]

### When did it happen?

09/11/2023 and days before

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

schwede70, 2023-11-10:
... additional information to the settings: feed from battery to grid is deactivated!
[image]

dirkjanfaber, 2023-11-10:
It looks like you are running v3.11, which does not support this functionality. The latest candidate versions however do. 

Updating can be done by setting the update feed to "candidate" and then updating the firmware.

schwede70, 2023-11-10:
Yes, this assumption was correct! I was running on v3.11. Just updated to v3.20~18. I will try with this version and provide feedback!
Thanks!

dirkjanfaber, 2023-11-10:
Thank you too. I'll close this issue then. We'll add an extra check to VRM that setting restrictions can only be done when running a capable version of Venus.

closed 2023-11-10 by dirkjanfaber