victronenergy/dynamic-ess #119: [VRM - Bug]: problem with days of low price difference

pokeplayer2, 2023-11-17:
### VRM portal ID

c0619ab4ae0e

### Country / region

Germany (de)

### B max

26

### TB max

6,5

### FB max

7

### TG max

30

### FG max

30

### Battery costs

0,02

### Buy price

(p+0.21)*1.19

### Sell price

0,115

### feed-in possible

yes

### feed-in control

no

### What happened?

Since a while I have the feeling that on a nearly flat price day the system not calculate for the customer and today is such a day.

### When did it happen?

17/11/2023 00:00

pokeplayer2, 2023-11-19:
next day same issue: 
[image]

dirkjanfaber, 2023-11-19:
It looks like this is caused by the "blaming the wrong consumption on the grid" that the system uses right now. There was less solar than expected and the system charged the difference from the grid.
Once again, we are aware of this and will come with a good solution. Until then, there is not much to do about it.