victronenergy/dynamic-ess #120: [VRM - Bug]: Dynamic ESS error code: No matching schedule available

h2009, 2023-11-21:
### Contact Details

layth@layth.co.uk

### VRM portal ID

48e7da8580e7

### Country / region

None

### feed-in possible

None

### feed-in control

None

### What happened?

below error still active, and batteries did not charge up correctly. 

Dynamic ESS error code: No matching schedule available	
	
This email is generated by Automatic Alarm Monitoring on the VRM Portal. More information:	
https://[www.victronenergy.com/live/vrm_portal:alarms#automatic_alarm_monitoring](https://www.victronenergy.com/live/vrm_portal:alarms#automatic_alarm_monitoring)	
	
https://[vrm.victronenergy.com/installation/224457/dashboard](https://vrm.victronenergy.com/installation/224457/dashboard)	
Site time: 2023-11-21 07:21:33 Europe/London

### When did it happen?

Site time: 2023-11-21 07:21:33 Europe/London	
[image]

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

dirkjanfaber, 2023-11-21:
We are aware of this and working on a fix. Thanks for reporting.

dirkjanfaber, 2023-11-21:
This has been fixed. All scheduling should have returned by now.

closed 2023-11-21 by dirkjanfaber