victronenergy/dynamic-ess #121: [VRM - Bug]: Dynamic ESS error code: No matching schedule available

pokeplayer2, 2023-11-21:
### VRM portal ID

c0619ab4ae0e

### Country / region

None

### B max

26

### TB max

6.5

### FB max

7

### TG max

30

### FG max

30

### Battery costs

0,02

### Buy price

(p+0.21)*1.19

### Sell price

0,115

### feed-in possible

yes

### feed-in control

no

### What happened?

Dynamic ESS error code: No matching schedule available: is a fresh error msg from today

### When did it happen?

21/11/2023 02:00

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

pokeplayer2, 2023-11-21:
It would be good to have better filter function at the Alarm page.

dirkjanfaber, 2023-11-21:
Closing this as duplicate from #120

closed 2023-11-21 by dirkjanfaber