victronenergy/dynamic-ess #125: [Feature]: Winter Algoritm needed

SimonYoungtree, 2023-12-21:
### Contact Details

siempoot@gmail.com

### Description

DESS is not making optimal use of low grid prices during winter. For example, we have a 7 hr window with very low prices (17,3 ct).
DESS however will not charge. I do not understand this. If you buy at 17,3 ct and add 15% inefficiency and 3 ct for wear and tear (of the battery) then the costs for these kwh's becomes ~23ct. So for any hours with prices higher than this it would be beneficial to take it from the battery. 

This is not what DESS is calculating (see graphs). DESS does not plan to charge. On the contrary it plans to discharge when prices are the lowest.

[image]

### Suggested Solution

Perhaps Victron is already studying on something. Until then I will use ESS during the winter months and use Node-Red to charge when prices are below 20ct, to pause the battery when prices are between 20ct and 28ct . And use the battery when prices are above. Plus sell back when battery SOC>50% and prices are >35ct. All by  manipulating SETPOINT and inserting SCHEDULED CHARGE LEVELS.

Also not 100% optimal, but better than DESS at the moment (I switch back to DESS to see what it would plan to compare regularly)
I think DESS is fairly good when the sun produces more KwH's.

(I am using Venus OS v3.13 and the latest Node-Red components)

### Alternatives

DESS will deal with it at some point in time.

dirkjanfaber, 2023-12-22:
Well, it did not plan to discharge to grid during those cheap buy hours, but use it for consumption instead. I do agree that more energy could have been taken from the grid during those hours. Tomorrow morning the prices are again at the same low stage and your system would have been able feed all (predicted) consumption from the battery until then.

SimonYoungtree, 2023-12-23:
I am looking for ways to influence the charging at very low prices, without stopping DESS. I have seen that you have at least 2 coping_strategies. When "1" the battery is idle and consumption is taken from the grid. When "2" the system follows the setpoint. In case of "2" I can manipulate the setpoint, making it to charge/discharge. 
Is there  a setting related to the copying startegy that I can change (I would this to be a variable in the DESS hourly schedule, but it is not)? And are there more than these 2 coping_strategies?

SimonYoungtree, 2024-02-26:
> There are 2 strategies right now, which are set automatically using the
> latest release + flow (0.1.14). I do advice to run the latest beta firmware
> (at the moment 3.30~9).

I assume you mean 3.30~7?

AbelLykens, 2024-02-27:
On update, does AC-OUT keep working?

dirkjanfaber, 2024-02-27:
Yes. At least, I don't see a reason why it should not.

closed 2024-02-27 by dirkjanfaber