victronenergy/dynamic-ess #127: sell to grid (still) has a to high priority, the charging of the batteries is oscillating

Edrettikketet, 2024-02-27:
This issue is a continuation of issue #115.
[image]
[image]
[image]
[image]

My SOC is set to 65%.
When the sun comes out, the MPPT starts loading the batteries until it is 67%.  Then it starts to push solar and battery power into the grid. When the charge level is back to 65% it starts loading the batteries again.  This all happens in the same hour (same sell price of the current).

It starts oscillating within the 2%.  How?

dirkjanfaber, 2024-02-27:
Did you import the new flow that adds the strategies and restrictions? Can you provide your site id?

Edrettikketet, 2024-02-27:
102c6b644cfd

dirkjanfaber, 2024-02-27:
I see that you are running firmware 3.21. Please use the latest beta feed for the strategies and restrictions to function.

You can switch via the remote console to this: _Settings -> Firmware -> Online updates -> Update feed_.

Edrettikketet, 2024-02-27:
This doesn't make a difference.  The batteries are uncharging again at 67%.
[image]

dirkjanfaber, 2024-02-27:
You still need to import the latest flow: https://github.com/victronenergy/dynamic-ess/blob/main/examples/fetch-dynamic-ess.json

Or do that from Node-RED:
[image]

AbelLykens, 2024-02-27:
[image]
I tried importing into a new flow, I got this error. Any thoughts?

I would really like some UPGRADING.md, because I have not used NR before and this is all new for me.

dirkjanfaber, 2024-02-27:
You need to install `node-red-dashboard` first, which you can do from the palette manager. I overlooked the need for that.

dirkjanfaber, 2024-02-27:
I've updated the [README](https://github.com/victronenergy/dynamic-ess/blob/main/README.md#error-message-when-importing-imported-unrecognised-types-ui_template-ui_group-and-ui_tab) to handle this. Btw the VRM implementation is really comparable to the Node-RED implementation. If you want to go for the more user friendly approach, I'd advice to go that route. You can read about that on https://community.victronenergy.com/articles/256073/dynamic-ess-on-beta-vrm-part-4.html

AbelLykens, 2024-02-27:
Thanks, that is a step in the right direction. After installing `node-red-dashboard` I still get:
> Flows stopped due to missing node types.
>    ui_template

[image]

I did this, first install node-red-dashboard.
[image]

Then I deleted the existing flow, then re-imported. Seems broken [?].

dirkjanfaber, 2024-02-27:
That `MODULE_NOT_FOUND` is unexpected. It looks like `node-red-dashboard` did not install properly. 

You can safely delete that `unkown: ui_template` node. But this is a bit strange.

Edrettikketet, 2024-02-27:
I have exactly the same problem.  How to remove these nodes and reinstall them again?

dirkjanfaber, 2024-02-27:
A click on the node, hitting the backspace button and deploying should be sufficient.

AbelLykens, 2024-02-27:
A restart seems to be required after installing `node-red-dashboard`. It now works here.

AbelLykens, 2024-02-27:
Wait no it does not:
[image]
But the flow now runs...

dirkjanfaber, 2024-02-27:
Can you click on the box left from the "Enable Dynamic ESS (mode: Node-RED)" inject node?
[image]
There should be a `4` below both orange nodes.

Edrettikketet, 2024-02-27:
> That `MODULE_NOT_FOUND` is unexpected. It looks like `node-red-dashboard` did not install properly.
> 
> You can safely delete that `unkown: ui_template` node. But this is a bit strange.

How can we correct this? Is there a way to delete these nodes and reload them?

dirkjanfaber, 2024-02-27:
I see that you have remote support enabled. If you don't mind, I can fix it remotely for you. Alternative is that you wait another day until I've finished describing on how to do it yourself.

Edrettikketet, 2024-02-27:
You may do it, but I always like to know how I can do it myself.  So you may also describe it tomorrow. (I have some patience ;-)

dirkjanfaber, 2024-02-27:
It turns out that it was just a matter of restarting Node-RED. No more "MODULE_NOT_FOUND" messages. It was less work that I had in mind to describe.

However I will describe the more thorough method [here](https://www.victronenergy.com/media/pg/Cerbo-S_GX/en/reset-to-factory-defaults-procedure.html) tomorrow.

dirkjanfaber, 2024-02-28:
You are welcome. Happy to help. 

You probably already know, but on https://community.victronenergy.com/ there are also a lot of knowledgeable and helpful people for all Victron products. In case of trouble, that is also an excellent place to ask questions.

I'll close this issue now as it has derailed a bit from the original problem. And, with the updates, I am fairly sure it has been resolved. If you still encounter the problem, just open a new issue.

closed 2024-02-28 by dirkjanfaber

Edrettikketet, 2024-03-07:
This comment was closed, but after updating as recommended the problem still persist.  The system is still oscillating between 55% (SoC) and 57% battery charge percentage.