victronenergy/dynamic-ess #128: BUG: Dynamic ESS not in Node-RED mode (undefined)

AbelLykens, 2024-02-27:
### Country / region

None

### feed-in possible

None

### feed-in control

None

### What happened?

[image]

I installed/imported the new flow. But it does not want to start.

[image]

The inject did not work:

[image]

### When did it happen?

2024-02-27 13:41 cet

dirkjanfaber, 2024-02-27:
And does manually clicking on that box next to "Enable Dynamic ESS (mode: Node_RED)" help?

AbelLykens, 2024-02-27:
No luck:
[image]

dirkjanfaber, 2024-02-27:
Not sure how I can reproduce this. What is your VRM or site id?

AbelLykens, 2024-02-27:
I'm `c0619ab4b555` :)

AbelLykens, 2024-02-27:
Now reverting to old firmware. Let's see if I get dess back.

AbelLykens, 2024-02-27:
Now on v3.22, no luck. It all seems kinda broken now (no dynamic things going on) . Not sure what steps to take.

dirkjanfaber, 2024-02-27:
I got it fixed on your site. For some reason the Victron Energy Client configuration node was not there. Added that one to your Node-RED installation. 

Now for me to figure out how that happened and prevent it from happening again.

dirkjanfaber, 2024-02-28:
Closing this issue. If it happens again, I'll investigate deeper.

closed 2024-02-28 by dirkjanfaber

Tomashaglund1, 2024-03-04:
> Victron Energy Client configuration

Hi, what node is "Victron Energy Client configuration node"? Most probably I'm having the same problem... Unable to activate dynamic ESSmode 4, it results in the error code on the right hand side in below screenshot.

[image]

dirkjanfaber, 2024-03-04:
Easiest solution is to drag another Victron node into the canvas, press deploy once and then it should appear.

Tomashaglund1, 2024-03-04:
Tried adding grid meter however it's stating "Unable to access the service endpoint."

[image]

Tomashaglund1, 2024-03-04:
Thanks got it working! :)