victronenergy/dynamic-ess #130: [Bug]: Noooo don't take from battery when prices are very low... ;)

AbelLykens, 2024-03-08:
### Country / region

None

### feed-in possible

None

### feed-in control

None

### What happened?

Very odd/unexpected. Prices currently are very low. So I started my washing machine and dishwasher.

But it seems to take power from the battery. That does not make a lot of sense, does it?

[image]

[image]

### When did it happen?

20240305T1301 CET

dirkjanfaber, 2024-03-08:
Just make sure. You are using the latest flow? The one that also sets restrictions and strategies?

AbelLykens, 2024-03-08:
Yes pretty sure :) Upgraded a week ago or so :)

[image]

AbelLykens, 2024-03-15:
[image]

It's currently delivering energy back, and next hour it plans to take from the net... Not ideal.

[image]

dirkjanfaber, 2024-03-15:
We did have some updates on this from 3.30~13 (which you are running right now) to 3.30~17 (latest right now). Can you update and check if the problem persits?

AbelLykens, 2024-03-18:
[image]
[image]
[image]
[image]

System is now getting in more solar than expected. Prices will rise in a bit, and the battery is not full. Would be nice if it charged rather than fed back 👀

dirkjanfaber, 2024-03-18:
Did you notice the new releases option to add extra (most of the time solar) energy into the battery to reduce the grid usage during the first part of the night. 

[image]

AbelLykens, 2024-03-18:
I must admit the `b_goal` does not (yet?) make much sense to me. Have not used it yet. But will set it to 50%/0 (as per your screenshots) and see what it does 😄

AbelLykens, 2024-03-18:
Right I see. I think that makes sense. Let's set it and see what happens. Would be useful if `goal_hour` would dynamically change based on expected sun I suppose. But this is cool already. Let's see what happens.

AbelLykens, 2024-03-19:
[image]
[image]
I updated to 0.1.17. And I set the variables. No luck yet, it seems. I kinda expect the blue line to go to at least 80% at 20:00?

Will now update the cerbo to `v3.30~21` and report back.

dirkjanfaber, 2024-03-19:
That is what I would have expected too. It should already work with the previous firmware.

AbelLykens, 2024-03-19:
No luck for me yet on `v3.30~21` with `0.1.17`.

[image]
[image]

AbelLykens, 2024-03-19:
Something happened! It's getting energy now. Cheers.

dirkjanfaber, 2024-04-10:
I'll close this issue now as it seems to be resolved.

closed 2024-04-10 by dirkjanfaber