victronenergy/dynamic-ess #131: [Feature]: Please refresh day-ahead prices more often

AbelLykens, 2024-03-08:
### Description

This is about local-hosted Node Red ESS.

### Suggested Solution

Please try getting dynamic prices maybe every 30min after 12:00. 
Often they are known eary, at 1230 for example. But then they might only be loaded at 15:00.

### Additional Context

I am not sure if my local nodered gets prices from ENTSOe or from VRM. If they come from the victronvrm api, maybe two decreases in refresh interval are needed: VRM getting them from "the market" and nodered getting them from VRM.

AbelLykens, 2024-03-08:
For example, today/now, my nodered/dess decided to charge at full capacity, though it's already known that price tomorrow are lower than today. But the system has not yet gotten this data 😉

AbelLykens, 2024-03-15:
Again today/now it's charging quite intensely, but tomorrow the prices are lower. It just has not loaded the prices yet..
[image]

dirkjanfaber, 2024-03-18:
The system already tried retrieving the data each hour of the day, but we increased that frequency from once an hour to 6 times an hour, which should make it respond faster.
Node-RED gets the info from VRM, VRM gets it from ENTSOE.

AbelLykens, 2024-03-18:
Nice! Both systems (my system + VRM) now try 6x/hr?

AbelLykens, 2024-03-19:
Let's see what happens. 12:50 prices are known. Let's hope they appear quickly 🔥

AbelLykens, 2024-03-19:
ℹ️   Last update: 3/19/2024, 1:00:49 PM 
No new prices it seems. Not sure how I can see if VRM has the new prices yet.

AbelLykens, 2024-03-19:
>we increased that frequency from once an hour to 6 times an hour

That's in some new release in the future maybe @dirkjanfaber ? 😉 . 
Because prices are known for 30mins now. And `6x/hr = */10`. 

I am looking at the higher prices tomorrow, see it's the lowest NOW (for the coming days) and am thinking: WHY DON'T YOU CHARGE LIKE CRAZY mr Victron machine 😄 .

[image]

dirkjanfaber, 2024-03-19:
Just heard that it was only modified for (beta) VRM users, not for Node-RED users. Having that adjusted for Node-RED users too, so it should be better tomorrow. Sorry about that and thanks for keeping a sharp eye on it.

AbelLykens, 2024-03-19:
Cheers @dirkjanfaber , appreciated. That means prices are loaded more often on (regular) VRM. And NodeRed would try hourly, or also `*/10` / `6x/hr`? 😉

dirkjanfaber, 2024-03-19:
Correct. Both implementation should be in sync tomorrow.

AbelLykens, 2024-03-20:
[image]
Let's see, 13:00, prices are in.

AbelLykens, 2024-03-20:
[image]

13:30, no new prices yet

AbelLykens, 2024-03-20:
(typing this as I go)

Prices could have been there at 13:10 I guess. But not there at:
` Last update: 3/20/2024, 1:42:37 PM`  (T+42min)
` Last update: 3/20/2024, 1:54:38 PM`  (T+54min)
` Last update: 3/20/2024, 2:06:38 PM`  (T+66min)
` Last update: 3/20/2024, 2:18:38 PM`  (T+78min)

_Just noticed it updates about every 12 minutes_

FYI @dirkjanfaber , it does not seem to load more quickly yet. Will keep my eye on it.

AbelLykens, 2024-03-20:
It just appeared (2+ hours later, ` Last update: 3/20/2024, 3:06:38 PM `). Something is not right yet.

AbelLykens, 2024-03-20:
ACK thanks!

AbelLykens, 2024-03-25:
Have my screen open again. Prices are in (13:00). Let's see when it all refreshes on my Node Red 🕺

AbelLykens, 2024-03-25:
Feels like it will refresh at 15:00 again. Not sure it was fixed @dirkjanfaber ...
Maybe re-open this issue? 🙂

dirkjanfaber, 2024-03-26:
Yes, lets do that.

AbelLykens, 2024-03-29:
Fixed!!

closed 2024-03-29 by AbelLykens

dirkjanfaber, 2024-03-29:
Ok, prices got just updated and processed in Node-RED (way before 15:00). Closing this issue.