victronenergy/dynamic-ess #132: [Bug]: /dess stopped working after update to v0.1.15

AbelLykens, 2024-03-15:
### Country / region

None

### feed-in possible

None

### feed-in control

None

### What happened?

I upgraded to v0.1.15 (palette manager --> click upgrade --> wait for upgrade to finish --> disable Node Red --> enable node red).
I then entered 0 and 0 for the b_goal vars.
Deployed and restarted flows. But now I am getting HTTP/504 on /dess.

Should I do something differently?

### When did it happen?

2021-03-15 16:55

### Relevant log output

```shell
Not sure where to get logs.
```

### Screenshots

```bash
![
```

AbelLykens, 2024-03-15:
[image]

Tried reinstalling and restarting. No luck. I am `c0619ab4b555` if you can take a look.

Also I see https://vrm-dynamic-ess-api.victronenergy.com/ returns HTTP/500, not sure if that is related.

AbelLykens, 2024-03-15:
Solved the "Not in ESS Mode" by following your previous hint:
https://github.com/victronenergy/dynamic-ess/issues/128#issuecomment-1977283049

still no /dess though ;)

dirkjanfaber, 2024-03-15:
What is your vrm id? I'll probably take a look at it later today.

AbelLykens, 2024-03-15:
Whoo magically /dess returned!

dirkjanfaber, 2024-03-15:
I fixed it on your site. Will push a new update out after dinner.

AbelLykens, 2024-03-15:
Eet smakelijk and thanks!

dirkjanfaber, 2024-03-15:
Thanks Got 0.1.16 out. For you there is not much need to update. Though it can't hurt.

closed 2024-03-15 by dirkjanfaber

Tomashaglund1, 2024-03-17:
Similiar problem here venus.local:1881/dess is not working after update from V0.1.14 --> 0.1.17 page renders --> "Node-RED unreachable". Dynamic ESS seems to be working tough.
Getting the error "Could not create a feasible schedule" after the 0.1.15 update most likely.
Ideas?

[image]