victronenergy/dynamic-ess #133: [Bug]: TypeError: Cannot read properties of undefined (reading '18')

gjongenelen, 2024-03-15:
### Contact Details

gino@jongenelen.net

### VRM portal ID

c0619ab55e90

### Country / region

Netherlands (nl)

### B max

4.8

### TB max

3

### FB max

3

### TG max

30

### FG max

30

### Battery costs

0.03

### Buy price

(p+0.13)*1.21+0.025

### Sell price

(p+0.13)*1.21+0.015

### feed-in possible

yes

### feed-in control

yes

### Version

0.1.15

### What happened?

Dess was running fine, but today the dashboard stopped loading. I found the following error:
15/03/2024, 18:43:33[node: prepare dess variables](https://172.16.42.104:1881/#)
function : (error)
"TypeError: Cannot read properties of undefined (reading '18')"

### When did it happen?

Now

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

dirkjanfaber, 2024-03-15:
Just pushed a 0.1.16 release out (with updated flows) that fixes this. Might be a few minutes before the palette manager knows about this update.

closed 2024-03-15 by dirkjanfaber

gjongenelen, 2024-03-15:
The error changed to:
```
15/03/2024, 19:44:27[node: prepare dess variables](https://myip:1881/#)
function : (error)
"TypeError: Cannot read properties of undefined (reading '19')"
```

dirkjanfaber, 2024-03-15:
Did you re-import the flow?

gjongenelen, 2024-03-15:
Yes, deleted and replaced the nodes on re-import

dirkjanfaber, 2024-03-15:
As VRM admin I've checked on your site what I overlooked (and fixed it). I see what that was now and I'll make another release. 

Meanwhile it should already work on your site.

gjongenelen, 2024-03-16:
Awesome! Dankjewel