victronenergy/dynamic-ess #135: [VRM - Bug]: can’t set DESS Minimum SOC

beccsjb, 2024-04-01:
### Contact Details

beccsjb@me.com

### VRM portal ID

d41243d2eed8

### Country / region

None

### B max

35

### TB max

2.7

### FB max

2.7

### TG max

6.5

### FG max

12

### Battery costs

0.1

### feed-in possible

yes

### feed-in control

yes

### What happened?

A bug happened!

I don’t seem to be able to change/alter the DESS Minimum SOC setting on the Cerbo GX

### When did it happen?

01/04/2024 20:00
[image]

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

dirkjanfaber, 2024-04-02:
That menu item is a remnant of the earlier implementation and will be removed completely from the remote console in a coming update.

closed 2024-04-02 by dirkjanfaber

beccsjb, 2024-04-02:
Ok, thank you. So where do I set the minimum SOC for the Dynamic ESS system them?

It seems that DESS isn’t charging from grid when power is free and instead charging when the grid cost is at its most expensive.