victronenergy/dynamic-ess #138: [Bug]: Unable to alter Green mode on setting .

MichaKersloot, 2024-04-23:
### Country / region

None

### feed-in possible

None

### feed-in control

None

### What happened?

I've unchecked the Green Mode, Deployed, but the system keeps charging as if the green_mode_on setting was still on. Looking in the Context Data green_mode_on still is true there:

[image]

### When did it happen?

23-04-2024 11:00

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

dirkjanfaber, 2024-04-23:
Yes, noticed it this morning too and working on an update. Probably to be released this afternoon.

Keeping this issue open until it has been released.

dirkjanfaber, 2024-04-23:
Yes, I'll add that.

closed 2024-04-23 by dirkjanfaber

MichaKersloot, 2024-04-24:
The setting seems to 'stick' now! But it looks like the calculations are still based on green_mode_on = true despite it setting to false. (id: c0619ab410ae )