victronenergy/dynamic-ess #139: [Feature]: Please add South Africa to the choices

MJS-001, 2024-04-23:
### Contact Details

mjs007mi4@gmail.com

### Description

Please add selection for South Africa (RSA) to the choices. Eskom provides TOU (time of use) tariffs. There is a peak, a standard and off peak energy charge. A time wheel correlate the time to the energy charges. It is different for weekdays (Mo-Fr), Saturdays and Sundays. Eskom has a low demand season which is every month except the three month of the high demand season (June, July August). The energy charge is increased one a year. The date for this increase is not the same for consumers. It depends on the financial year of Eskom and the energy resellers.

### Suggested Solution

Split the fixed weekend schedule in one schedule for Saturdays and one schedule for Sundays for buying and selling.
Implement more data-sets so that energy costs for different times of the year can be stored. In RSA this would need a second data set to accommodate high demand and low demand seasons. 

[2023-24 Electricity Tariffs.pdf](https://github.com/victronenergy/dynamic-ess/files/15076519/2023-24.Electricity.Tariffs.pdf)

[20231106 BDE Fancourt Tariffs Letter 2023.pdf](https://github.com/victronenergy/dynamic-ess/files/15076537/20231106.BDE.Fancourt.Tariffs.Letter.2023.pdf)

dirkjanfaber, 2024-04-28:
On the VRM implementation you can configure this already when you set the prices to "fixed". Though, when I read about the different seasons, it will probably need you to adjust the prices there every change of season.

MJS-001, 2024-04-28:
While settling it up i came across these issues.
The Saturday Sunday issue can't be resolved without providing an additional set of data. One can imagine that other utilities have different flat prices for every day of the week. In that case one would need 7 sets of data.
The change of seasons can be overcome by editing the data sets multiple times per year. You will admit that this is inconvenient. Could you make it possible to read the data from a spreadsheet? That would help in the administration of multiple sites.