victronenergy/dynamic-ess #140: [Bug]: System stuck in Green mode

MichaKersloot, 2024-04-26:
### VRM portal ID

c0619ab410ae

### Country / region

None

### feed-in possible

None

### feed-in control

None

### Version

0.1.20

### What happened?

System is stuck in Green mode since it has been implemented/updated I can see the mode changes in the raw data, but the setup never discharges anymore. 

[image]

### When did it happen?

26/04/2024 12:00

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

MichaKersloot, 2024-04-26:
Hmm.. after the bug-report I've rechecked the settings and found 'tg max' to be reset to 0. Maybe that is a bug in itself, but has nothing to do with the green mode setting.

dirkjanfaber, 2024-04-26:
That is good to know. You can also check which parameters get send when having the used url shown in the debug tab.

MichaKersloot, 2024-04-26:
I don't think I'm able to reproduce this situation, so I'll close this.

closed 2024-04-26 by MichaKersloot