victronenergy/dynamic-ess #141: [Bug]: Dynamic ESS option dissappeared from Console ESS menu

RobMeerwijk, 2024-04-26:
### Contact Details

rob@robsadvies.nl

### VRM portal ID

c0619ab30da0

### Country / region

Netherlands (nl)

### B max

32

### TB max

8

### FB max

8

### TG max

17

### FG max

17

### Battery costs

.02

### feed-in possible

yes

### feed-in control

None

### Version

latest (venus os 3.31)

### What happened?

I picked "external control" as ESS mode. I returned to "optimized (with battery life)" After this switch the option "dynamic ess" was gone from remote console. VRM still shows normal behaviour. The schedule does show selling and buying in the hours and day ahead. I got two emails. The first warned that the settings were invalid for DESS. The second made clear that the correct settings were restored. 

### When did it happen?

26.04.2024 22:33

### Relevant log output

```shell
First mail:

Dynamic ESS error code: ESS mode incorrect	
	
This email is generated by Automatic Alarm Monitoring on the VRM Portal. More information:	
https://www.victronenergy.com/live/vrm_portal:alarms#automatic_alarm_monitoring	
	
https://vrm.victronenergy.com/installation/421456/dashboard	
Site time: 2024-04-26 22:32:54 Europe/Berlin	
	
Second mail:

Dynamic ESS error code changed from 'ESS mode incorrect' to 'No error'	
	
This email is generated by Automatic Alarm Monitoring on the VRM Portal. More information:	
https://www.victronenergy.com/live/vrm_portal:alarms#automatic_alarm_monitoring	
	
https://vrm.victronenergy.com/installation/421456/dashboard	
Site time: 2024-04-26 22:33:25 Europe/Berlin
```

dirkjanfaber, 2024-04-27:
What happens when you switch the toggle in the controls section of the dashboard? See screenshot:
[image]

RobMeerwijk, 2024-04-27:
Solved! Thank you. So not a bug after all. I was not thinking about that option. I had switched DESS on and off onder Settings and Dynamic ESS. In fact. It was switched on under Settings and switched off under Controls.

Thank you for your support.

Rob Meerwijk

closed 2024-04-27 by RobMeerwijk