victronenergy/dynamic-ess #143: [Bug]: Time discreprency in DESS

Tomashaglund1, 2024-05-03:
### Contact Details

tomashaglund1@gmail.com

### VRM portal ID

c0619ab38161

### Country / region

Finland (fi)

### B max

15

### TB max

1.8

### FB max

2.4

### TG max

4.7

### FG max

4.7

### Battery costs

0.006

### Buy price

(p+0.0428+0.0279372)*1.24

### Sell price

(p-0.0045)

### feed-in possible

yes

### feed-in control

yes

### Version

0.1.20

### What happened?

There's a discreprency in the NTP vs. the time displayed in DESS. What could be causing this?

### When did it happen?

It's happening all the time

[image]

Multiplus 2 time

[image]

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

dirkjanfaber, 2024-05-14:
On the screenshots it looks like the system thinks it needs to draw 48 hours, but only does draw 24 hours. Which causes the grey bar to be drawn halfway where it should. 
Is that also the case if you refresh the screen? And which browser are you using?

Tomashaglund1, 2024-05-17:
Google chrome ver. 125.0.6422.61 & Issue still there after refreshing & Issue is also there if running incognito window.

e.g.
On Monday when the spotprices are updated at 14:15 and /DESS refreshes with 48h timescale it's showing ok untill Tuesday 00:00 when it refresh to 24h timescale then there's a time discreprency of 4h when local time is 08:00 /DESS is showing 04:00