victronenergy/dynamic-ess #146: [Feature]: Add Australia to location list

moto501, 2024-06-12:
### Contact Details

evrose@cisco.com

### Description

Australia has dynamic wholesale pricing. Pricing available through API via AEMO https://dev.aemo.com.au/

### Suggested Solution

Add Australia to the list of location options. 

### Alternatives

None

### Additional Context

None

dirkjanfaber, 2024-06-12:
I took a quick look at that link and it seems you need to register for that API before being able to access it.
Just to get a clear image before digging any deeper into it. Are the dynamic prices also changing hourly in Australia?

And any info on how popular Dynamic pricing usage is in Australia?

moto501, 2024-06-12:
Hi,

Prices change ever 5 mins but we typically use 30min pricing.

We have periods of price spikes both in terms of buy and feed-in (sell).

Victron with zero-feed-in works well to prevent feed-in, but often we want to buy/sell dynamically.  I saw your flow wave thought this would be perfect.

Is there anything I cash do to help?

moto501, 2024-06-12:
Also,  dynamic pricing is gaining popularity due to the uptick of battery installations.  Some providers offer this with their 'approved' batteries,  but the user loses control and Victron ESS isn't supported.

moto501, 2024-06-12:
Another option is to link directly to the Amber Electric API. Amber is a wholesale energy provider for the Australian consumer. They provide API access as default. With this, a hook to their API would provide real-time pricing. I would be an easier option that providing a general API link to the AEMO. There is a Home-Assistant integration (HACS) already which has the API hooks.

dirkjanfaber, 2024-06-15:
What we currently do is to download the prices once they become available and store them in our own databases. From there we do the schedule calculation for all of the sites.
If you can read the prices from the API of your provider, you can also add them as "fixed" prices to the VRM implementation. On https://community.victronenergy.com/articles/293324/node-red-vrm-api-and-dynamic-ess-1.html I've posted an article describing how you can get started accomplishing that.
At the moment, I'd say that is the quickest alternative solution for you. Once it works for you and you share your flow, it will make it easier for us to add Australia natively to VRM too.

moto501, 2024-06-15:
Will do. Thanks :)