victronenergy/dynamic-ess #147: [VRM - Bug]: dynamic ess menu not visible anymore

Wim275, 2024-06-15:
### Contact Details

wim@clinckspoor.be

### VRM portal ID

c0619ab4a1a9

### Country / region

Belgium (be)

### B max

8

### TB max

4

### FB max

4

### TG max

15

### FG max

22

### Battery costs

2.7

### feed-in possible

None

### feed-in control

None

### What happened?

A bug happened!
For some reason, the ess setting menu disappeared in the navigation of the vrm. In the dashboard I can still swith ess on and off.

### When did it happen?

Don’t realy know

schlotze, 2024-07-23:
I have the same problem.
I am running venus os v3.40

Wim275, 2024-07-23:
I have 3.13, but nobody from victron seems to pick up this problem, are there other ways to contact victron support ?
Wim.

mpvader, 2024-07-23:
Hey both, I’ll make sure someone looks into this tomorrow / check what is being done on this.

mpvader, 2024-07-23:
> A bug happened!
> For some reason, the ess setting menu > disappeared in the navigation of the vrm. In the > dashboard I can still swith ess on and off.

You mean the dynamic ess menu in VRM, right?

> I have 3.13

Venus OS v3.13 is quite old; for any trouble shooting; first thing to do will be to update it to the latest version; which at the moment is Venus OS v3.40

Wim275, 2024-07-24:
Thanks for your answer, yes the dynamic ess menu is not there anymore. 
[image]
[image]

Wim275, 2024-07-24:
I’m on holliday now, I will do the os firmware upgrade on the 3th of august.

closed 2024-07-24 by Wim275