victronenergy/dynamic-ess #148: [Bug]: System started? to oscillate since 11th of june

MichaKersloot, 2024-06-17:
### VRM portal ID

c0619ab410ae

### Country / region

None

### feed-in possible

None

### feed-in control

None

### What happened?

Just noticed this weekend because of the excessive noise during the night. But looking in the history my system started to oscillate since the 11th of june. See for example
[image]

### When did it happen?

Good example is Jun 15th or Jun 12th

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

dirkjanfaber, 2024-06-17:
Where you on v3.31 all that time? I do suggest first updating to v3.33.

MichaKersloot, 2024-06-17:
I'll do...... Communication between Pylontech en Victron also lost since 10:45. So I'll update and reboot and hope it all resolves.

MichaKersloot, 2024-06-19:
Ok, communication is fixed, but still oscillation:
[image]

dirkjanfaber, 2024-06-20:
Can you set the vrm log interval to each minute instead of every 15 minutes? You can find that setting in the remote console under _Settings -> VRM online portal_.

MichaKersloot, 2024-06-20:
Yes, i've done now.

MichaKersloot, 2024-06-24:
Hi, are you able to find the issue? It also seems the short peaks and valleys just before the end of the hour are back. Maybe some dampening could be a solution?