victronenergy/dynamic-ess #149: [VRM - Bug]: dynamic prices do not correspond to ectual prices

Edrettikketet, 2024-06-17:
### Contact Details

e.peetam@hotmail.com

### VRM portal ID

102c6b644cfd

### Country / region

Netherlands (nl)

### B max

22.8

### TB max

5

### FB max

5

### TG max

5.75

### FG max

5.75

### Battery costs

0.06

### Buy price

(p+0.0175+0.1088)*1.21

### Sell price

(p+0.0175+0.1088)*1.21

### feed-in possible

yes

### feed-in control

yes

### What happened?

A bug happened!

How is this possible?  There seems a mismatch between the actual hour rates and the VRM rates.  
How?
[image]

### When did it happen?

DD/MM/YYYY HH:MM

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

devilmastah, 2024-06-26:
I think this answers my confusion this morning when i saw this:

[image]
[image]

yesterday seems to be correct however so i do not fully get how this happens:

[image]
[image]

I am not sure about the formula here, i use this, which looked correct so far?

[
[image]
](url)

dirkjanfaber, 2024-06-26:
For today's prices, there was a partial decoupling of some European countries from the single Day-ahead market.

For more background see
https://www.icis.com/explore/resources/news/2024/06/25/11011378/day-ahead-power-price-spreads-soar-as-markets-decouple-on-technical-issue/

For today it is probably best to either accept the calculation with the (wrong) prices or disable Dynamic ESS for a few hours and do your steering manually.
I am not sure how all different providers handle this case.

For the longer term we will implement a better solution.

For the actual issue (which I've overlooked so far), we will take a closer look as well.