victronenergy/dynamic-ess #150: [VRM - Bug]: system doesn't follow the VRM schedule

Edrettikketet, 2024-07-08:
### Contact Details

e.peetam@hotmail.com

### VRM portal ID

102c6b644cfd

### Country / region

Netherlands (nl)

### B max

22.8

### TB max

5

### FB max

5

### TG max

5.75

### FG max

5.75

### Battery costs

0.06

### Buy price

(p+0.0175+0.1088)*1.21

### Sell price

(p+0.0175+0.1088)*1.21

### feed-in possible

yes

### feed-in control

yes

### What happened?

A bug happened!
Many times the feed in is scheduled in the VRM schedule, but not obeyed by the cerbo.  How?
This is the schedule:
[image]
But the feed-in function is off:
[image]
What to do to correct this, or is this a bug?

### When did it happen?

DD/MM/YYYY HH:MM

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

Edrettikketet, 2024-07-12:
Apparently the above has to do with the error: Dynamic ESS error code: No matching schedule available.
After a reboot the message is gone and the alarm is cleared, but the system doesn't follow the schedule. The next day the error appears again.  How to repair this?

Solear, 2024-08-03:
I've got the same error (RPI 4 8GB, Venus OS large). Node-red gui shows correct schedules. But VRM shows "error: Dynamic ESS error code: No matching schedule available".
Timezone and time-setting is correct and the same(VRM and venus os).
Reboot or fresh installation oder wait 1-2 days doesnt help.

But, the scheduler in node-red (not node red gui) shows other times...?
[image]
[image]