victronenergy/dynamic-ess #157: [Bug]: Is VRM connection down?

Tomashaglund1, 2024-10-05:
### Contact Details

tomashaglund1@gmail.com

### VRM portal ID

c0619ab38161

### Country / region

Finland (fi)

### B max

16.5

### TB max

1.8

### FB max

2.4

### TG max

4.7

### FG max

4.7

### Battery costs

0.05

### Buy price

(p+0.0428+0.0279372)*1.255

### Sell price

(p-0.0045)

### feed-in possible

yes

### feed-in control

yes

### Version

0.1.23

### What happened?

A bug happened!
Have been away from home 4 days and noticed this when i came back home
Seems like there's no connection to VRM?

### When did it happen?

05/10/2024 00:00

### Relevant log output

```shell
05/10/2024, 13:12:13node: Dynamic ESS
msg : string[46]
"Unable to connect to VRM for more than an hour"
05/10/2024, 13:12:18node: prepare dess variables
function : (error)
"TypeError: Cannot read properties of undefined (reading 'toFixed')"
05/10/2024, 13:12:22node: prepare dess variables
function : (error)
"TypeError: Cannot read properties of undefined (reading 'toFixed')"
05/10/2024, 13:12:43node: prepare dess variables
function : (error)
"TypeError: Cannot read properties of undefined (reading 'toFixed')"
05/10/2024, 13:13:13node: Dynamic ESS
msg : string[46]
"Unable to connect to VRM for more than an hour"
05/10/2024, 13:13:18node: Dynamic ESS
msg : Object
{ url: "https://vrm-dynamic-ess-api.vi…", options: object, headers: object }
05/10/2024, 13:13:19node: Dynamic ESS
msg : error
"AxiosError: Request failed with status code 500"
05/10/2024, 13:13:25node: prepare dess variables
function : (error)
"TypeError: Cannot read properties of undefined (reading 'toFixed')"
05/10/2024, 13:13:46node: prepare dess variables
function : (error)
"TypeError: Cannot read properties of undefined (reading 'toFixed')"
05/10/2024, 13:14:13node: Dynamic ESS
msg : string[46]
"Unable to connect to VRM for more than an hour"
05/10/2024, 13:14:28node: prepare dess variables
function : (error)
"TypeError: Cannot read properties of undefined (reading 'toFixed')"
05/10/2024, 13:14:49node: prepare dess variables
function : (error)
"TypeError: Cannot read properties of undefined (reading 'toFixed')"
05/10/2024, 13:15:13node: Dynamic ESS
msg : string[46]
"Unable to connect to VRM for more than an hour"
05/10/2024, 13:15:31node: prepare dess variables
function : (error)
"TypeError: Cannot read properties of undefined (reading 'toFixed')"
05/10/2024, 13:15:52node: prepare dess variables
function : (error)
"TypeError: Cannot read properties of undefined (reading 'toFixed')"
05/10/2024, 13:16:13node: Dynamic ESS
msg : string[46]
"Unable to connect to VRM for more than an hour"
05/10/2024, 13:16:20node: prepare dess variables
function : (error)
"TypeError: Cannot read properties of undefined (reading 'toFixed')"
05/10/2024, 13:16:55node: prepare dess variables
function : (error)
"TypeError: Cannot read properties of undefined (reading 'toFixed')"
05/10/2024, 13:17:13node: Dynamic ESS
msg : string[46]
"Unable to connect to VRM for more than an hour"
05/10/2024, 13:17:23node: prepare dess variables
function : (error)
"TypeError: Cannot read properties of undefined (reading 'toFixed')"
05/10/2024, 13:17:59node: prepare dess variables
function : (error)
"TypeError: Cannot read properties of undefined (reading 'toFixed')"
05/10/2024, 13:18:13node: Dynamic ESS
msg : string[46]
"Unable to connect to VRM for more than an hour"
05/10/2024, 13:19:02node: prepare dess variables
function : (error)
"TypeError: Cannot read properties of undefined (reading 'toFixed')"
05/10/2024, 13:19:13node: Dynamic ESS
msg : string[46]
"Unable to connect to VRM for more than an hour"
05/10/2024, 13:20:05node: prepare dess variables
function : (error)
"TypeError: Cannot read properties of undefined (reading 'toFixed')"
05/10/2024, 13:20:13node: Dynamic ESS
msg : string[46]
"Unable to connect to VRM for more than an hour"
05/10/2024, 13:21:13node: Dynamic ESS
msg : string[46]
"Unable to connect to VRM for more than an hour"
05/10/2024, 13:22:13node: Dynamic ESS
msg : string[46]
"Unable to connect to VRM for more than an hour"
05/10/2024, 13:23:13node: Dynamic ESS
msg : string[46]
"Unable to connect to VRM for more than an hour"
05/10/2024, 13:24:13node: Dynamic ESS
msg : string[46]
"Unable to connect to VRM for more than an hour"
05/10/2024, 13:25:13node: Dynamic ESS
msg : string[46]
"Unable to connect to VRM for more than an hour"
05/10/2024, 13:25:18node: Dynamic ESS
msg : Object
{ url: "https://vrm-dynamic-ess-api.vi…", options: object, headers: object }
05/10/2024, 13:25:19node: Dynamic ESS
msg : error
"AxiosError: Request failed with status code 500"
05/10/2024, 13:26:13node: Dynamic ESS
msg : string[46]
"Unable to connect to VRM for more than an hour"
```

### Screenshots

```bash
![DESCRIPTION](LINK.png)
```

Tomashaglund1, 2024-10-05:
[image]

Tomashaglund1, 2024-10-07:
Strangely it started to work by itself or by a "helping hand" from Victron side as I did nothing...
Closing the issue!
[image]

closed 2024-10-07 by Tomashaglund1

dirkjanfaber, 2024-10-07:
Fwiw: It was caused by a problem with retrieval of the prices from Entso-E, which started on Friday and also happend last night. After fixing it manually twice, we are adding some extra checks to have this kind of problem recover by itself.