victronenergy/dynamic-ess #163: [Bug]: The pricing is wrong since a few days.

andreashackel, 2024-10-16:
### Contact Details

ankawe@gmx.net

### VRM portal ID

48e7da85cff9

### Country / region

Germany (de)

### B max

10.5

### TB max

4

### FB max

4

### TG max

0

### FG max

0

### Battery costs

0.04

### Buy price

p

### Sell price

0.12

### feed-in possible

no

### feed-in control

no

### Version

0.1.20

### What happened?

The pricing is wrong since a few days.
I compared the prices with tibber and entsoe.
I noticed in the price graphic that there is a ? instead of €. “Price in ?/kWh”

### When did it happen?

16/10/2024 00:00 to 17/10/2024 23:59
 
Example:

16/10/2024 15:00 
p_b: 15: 0.06355
entsoe: 16/10/2024 15:00 - 16/10/2024 15:15	14.18 -> 0,01418€/kWh

16/10/2024 17:00 
p_b: 17: 0.1235999999999999
entsoe: 16/10/2024 17:00 - 16/10/2024 17:15	72.95 -> 0,07295€/kWh

### Relevant log output

```shell
16.10.2024 15:46 Object p_b
0: 0.01704
1: 0.01215
2: 0.01048
3: 0.00561
4: 0.01001
5: 0.0298
6: 0.07051
7: 0.10139
8: 0.09253
9: 0.07376
10: 0.06222
11: 0.05202
12: 0.04654
13: 0.04752
14: 0.0600399999999999
15: 0.06355
16: 0.09268
17: 0.1235999999999999
18: 0.18621
19: 0.17219
20: 0.1175
21: 0.09781
22: 0.0890499999999999
23: 0.0822099999999999
24: 0.0371
25: 0.03525
26: 0.03517
27: 0.03436
28: 0.03933
29: 0.0622
30: 0.09874
31: 0.1328899999999999
32: 0.15361
33: 0.11794
34: 0.10095
35: 0.09553
36: 0.08836
37: 0.0825999999999999
38: 0.0741599999999999
39: 0.08708
40: 0.10579
41: 0.15959
42: 0.2571
43: 0.26974
44: 0.158
45: 0.11412
46: 0.09966
47: 0.0948

https://newtransparency.entsoe.eu/market/energyPrices?appState=%7B%22sa%22%3A%5B%22BZN%7C10Y1001A1001A82H%22%5D%2C%22st%22%3A%22BZN%22%2C%22mm%22%3Atrue%2C%22ma%22%3Afalse%2C%22sp%22%3A%22HALF%22%2C%22dt%22%3A%22TABLE%22%2C%22df%22%3A%222024-10-16%22%2C%22tz%22%3A%22CET%22%7D

MTU	Sequence 1	Sequence 2	Day-ahead
(EUR/MWh)	Day-ahead
(EUR/MWh)	
16/10/2024 00:00 - 16/10/2024 00:15	23.35	49.10			
16/10/2024 00:15 - 16/10/2024 00:30	23.35	37.30			
16/10/2024 00:30 - 16/10/2024 00:45	23.35	19.60			
16/10/2024 00:45 - 16/10/2024 01:00	23.35	7.59			
16/10/2024 01:00 - 16/10/2024 01:15	12.80	41.80			
16/10/2024 01:15 - 16/10/2024 01:30	12.80	19.58			
16/10/2024 01:30 - 16/10/2024 01:45	12.80	10.09			
16/10/2024 01:45 - 16/10/2024 02:00	12.80	5.10			
16/10/2024 02:00 - 16/10/2024 02:15	5.30	28.45			
16/10/2024 02:15 - 16/10/2024 02:30	5.30	12.59			
16/10/2024 02:30 - 16/10/2024 02:45	5.30	10.09			
16/10/2024 02:45 - 16/10/2024 03:00	5.30	7.59			
16/10/2024 03:00 - 16/10/2024 03:15	3.61	10.10			
16/10/2024 03:15 - 16/10/2024 03:30	3.61	10.09			
16/10/2024 03:30 - 16/10/2024 03:45	3.61	10.09			
16/10/2024 03:45 - 16/10/2024 04:00	3.61	18.01			
16/10/2024 04:00 - 16/10/2024 04:15	3.16	0.03			
16/10/2024 04:15 - 16/10/2024 04:30	3.16	9.79			
16/10/2024 04:30 - 16/10/2024 04:45	3.16	15.04			
16/10/2024 04:45 - 16/10/2024 05:00	3.16	25.06			
16/10/2024 05:00 - 16/10/2024 05:15	8.20	-27.46			
16/10/2024 05:15 - 16/10/2024 05:30	8.20	5.01			
16/10/2024 05:30 - 16/10/2024 05:45	8.20	31.47			
16/10/2024 05:45 - 16/10/2024 06:00	8.20	74.93			
16/10/2024 06:00 - 16/10/2024 06:15	42.59	0.09			
16/10/2024 06:15 - 16/10/2024 06:30	42.59	45.00			
16/10/2024 06:30 - 16/10/2024 06:45	42.59	69.50			
16/10/2024 06:45 - 16/10/2024 07:00	42.59	88.37			
16/10/2024 07:00 - 16/10/2024 07:15	67.29	48.60			
16/10/2024 07:15 - 16/10/2024 07:30	67.29	62.21			
16/10/2024 07:30 - 16/10/2024 07:45	67.29	89.90			
16/10/2024 07:45 - 16/10/2024 08:00	67.29	97.41			
16/10/2024 08:00 - 16/10/2024 08:15	66.52	111.09			
16/10/2024 08:15 - 16/10/2024 08:30	66.52	76.80			
16/10/2024 08:30 - 16/10/2024 08:45	66.52	62.40			
16/10/2024 08:45 - 16/10/2024 09:00	66.52	43.60			
16/10/2024 09:00 - 16/10/2024 09:15	54.29	114.20			
16/10/2024 09:15 - 16/10/2024 09:30	54.29	68.60			
16/10/2024 09:30 - 16/10/2024 09:45	54.29	45.72			
16/10/2024 09:45 - 16/10/2024 10:00	54.29	-4.91			
16/10/2024 10:00 - 16/10/2024 10:15	26.68	76.00			
16/10/2024 10:15 - 16/10/2024 10:30	26.68	25.83			
16/10/2024 10:30 - 16/10/2024 10:45	26.68	5.08			
16/10/2024 10:45 - 16/10/2024 11:00	26.68	-9.93			
16/10/2024 11:00 - 16/10/2024 11:15	14.16	55.70			
16/10/2024 11:15 - 16/10/2024 11:30	14.16	13.21			
16/10/2024 11:30 - 16/10/2024 11:45	14.16	0.09			
16/10/2024 11:45 - 16/10/2024 12:00	14.16	-9.91			
16/10/2024 12:00 - 16/10/2024 12:15	4.09	15.09			
16/10/2024 12:15 - 16/10/2024 12:30	4.09	5.10			
16/10/2024 12:30 - 16/10/2024 12:45	4.09	8.02			
16/10/2024 12:45 - 16/10/2024 13:00	4.09	-2.40			
16/10/2024 13:00 - 16/10/2024 13:15	2.04	2.52			
16/10/2024 13:15 - 16/10/2024 13:30	2.04	2.53			
16/10/2024 13:30 - 16/10/2024 13:45	2.04	-1.07			
16/10/2024 13:45 - 16/10/2024 14:00	2.04	10.02			
16/10/2024 14:00 - 16/10/2024 14:15	4.00	-14.99			
16/10/2024 14:15 - 16/10/2024 14:30	4.00	0.02			
16/10/2024 14:30 - 16/10/2024 14:45	4.00	10.02			
16/10/2024 14:45 - 16/10/2024 15:00	4.00	22.53			
16/10/2024 15:00 - 16/10/2024 15:15	14.18	-7.46			
16/10/2024 15:15 - 16/10/2024 15:30	14.18	5.04			
16/10/2024 15:30 - 16/10/2024 15:45	14.18	25.06			
16/10/2024 15:45 - 16/10/2024 16:00	14.18	60.06			
16/10/2024 16:00 - 16/10/2024 16:15	54.92	-12.41			
16/10/2024 16:15 - 16/10/2024 16:30	54.92	40.13			
16/10/2024 16:30 - 16/10/2024 16:45	54.92	72.30			
16/10/2024 16:45 - 16/10/2024 17:00	54.92	92.30			
16/10/2024 17:00 - 16/10/2024 17:15	72.95	25.59			
16/10/2024 17:15 - 16/10/2024 17:30	72.95	67.41			
16/10/2024 17:30 - 16/10/2024 17:45	72.95	90.20			
16/10/2024 17:45 - 16/10/2024 18:00	72.95	116.50			
16/10/2024 18:00 - 16/10/2024 18:15	84.47	70.31			
16/10/2024 18:15 - 16/10/2024 18:30	84.47	87.40			
16/10/2024 18:30 - 16/10/2024 18:45	84.47	93.80			
16/10/2024 18:45 - 16/10/2024 19:00	84.47	82.10			
16/10/2024 19:00 - 16/10/2024 19:15	88.38	94.93			
16/10/2024 19:15 - 16/10/2024 19:30	88.38	92.41			
16/10/2024 19:30 - 16/10/2024 19:45	88.38	87.48			
16/10/2024 19:45 - 16/10/2024 20:00	88.38	52.59			
16/10/2024 20:00 - 16/10/2024 20:15	65.31	109.23			
16/10/2024 20:15 - 16/10/2024 20:30	65.31	83.40			
16/10/2024 20:30 - 16/10/2024 20:45	65.31	42.60			
16/10/2024 20:45 - 16/10/2024 21:00	65.31	30.09			
16/10/2024 21:00 - 16/10/2024 21:15	61.86	86.60			
16/10/2024 21:15 - 16/10/2024 21:30	61.86	72.10			
16/10/2024 21:30 - 16/10/2024 21:45	61.86	31.70			
16/10/2024 21:45 - 16/10/2024 22:00	61.86	17.58			
16/10/2024 22:00 - 16/10/2024 22:15	35.14	77.45			
16/10/2024 22:15 - 16/10/2024 22:30	35.14	61.20			
16/10/2024 22:30 - 16/10/2024 22:45	35.14	46.05			
16/10/2024 22:45 - 16/10/2024 23:00	35.14	10.09			
16/10/2024 23:00 - 16/10/2024 23:15	26.54	67.41			
16/10/2024 23:15 - 16/10/2024 23:30	26.54	39.55			
16/10/2024 23:30 - 16/10/2024 23:45	26.54	10.10			
16/10/2024 23:45 - 17/10/2024 00:00	26.54	2.58
```

andreashackel, 2024-10-17:
I checked the pricing today and everything is fine.

dirkjanfaber, 2024-10-18:
We looked into it and found, and fixed, a bug, regarding this issue.

closed 2024-10-18 by dirkjanfaber